janelia-arduino/TMC2209 #87: TMC2209 V1.3 offers 2 pins RX and TX instead of only one UART pin

hefl99, 2025-02-19:
The communication section of the Readme only covers older TMC2209 models with one UART pin.

However newer models such as the V1.3 have 2 pins: RX and TX

Could you please provide an example on how to wire the TMC2209 V1.3 to an Arduino Uno R4 Minima?

Thanks a lot in advance :)

hefl99, 2025-02-19:
[BIGTREETECH TMC2209 V1.3 User Manual.pdf](https://github.com/user-attachments/files/18867159/BIGTREETECH.TMC2209.V1.3.User.Manual.pdf)

peterpolidoro, 2025-02-19:
You had me all excited when I read the title of this issue. I have been wishing that the TMC2209 would come out with a new chip that had two independent pins, TX and RX, instead of the odd one wire UART interface. Your link is to a board that uses the TMC2209 chip with the one wire interface, but breaks that out that one pin on the chip to two pins on the board. I do not see a schematic in the manual, but it appears you need to solder in a resistor to connect both board pins to the one on the chip. So the circuit probably looks like the one included in my documentation where a resistor separates the TX and RX lines, but they are not actually independent pins like UART is on most chips.

hefl99, 2025-02-19:
So you are saying even though on the silkscreen of the TMC2209 chipboard it reads RX and TX at two different pins, it is indeed only one pin to the chip?
The silkscreen is very irritating then…
Thank you anyway for your quick reply

peterpolidoro, 2025-02-19:
I am saying that you should check out the schematic of the board to see for sure how it is wired. I do not know of any TMC2209 chips with separate TX and RX lines, so my guess is that yes, both board pins probably connect to the single pin on the chip with a resistor separating them.

closed 2025-02-25 by hefl99