janelia-arduino/TMC2209 #88: Library incompatible with Uno R4

hefl99, 2025-02-24:
When uploading a sketch using the TMC2209 library, the Uno R4 disconnects and freezes. Only a complete reboot and removing the sketch brings it back. It seems to me that this library is not compatible with the Arduino Uno R4.
@peterpolidoro, could you please update the library to work with the Renesas RA4M1 chip of the Uno R4?

hefl99, 2025-02-25:
I tried updating the library. Now the Uno R4 is no longer disconnecting and freezing.
However, still there is no communication.

I connected:
Arduino -> TMC2209
5V -> VIO
GND -> GND
TX -> 1kOhm ---I
                           I 
                          V
RX ----------->  UART

I tried a very simple sketch first, but i only get zeros as answers:
```
#include <TMC2209.h>

HardwareSerial & serial_stream = Serial1;

TMC2209 stepper_driver;

void setup() {
  Serial.begin(9600);
  delay(1000);

  stepper_driver.setup(serial_stream, 9600, TMC2209::SERIAL_ADDRESS_0);
}

void loop() {
  
  Serial.print("getVersion?: ");
  Serial.println(stepper_driver.getVersion());

  Serial.print("Is communicating?: ");
  Serial.println(stepper_driver.isCommunicating());

  Serial.print("isSetupAndCommunicating?: ");
  Serial.println(stepper_driver.isSetupAndCommunicating());

  Serial.print("isCommunicatingButNotSetup?: ");
  Serial.println(stepper_driver.isCommunicatingButNotSetup());

  Serial.print("hardwareDisabled?: ");
  Serial.println(stepper_driver.hardwareDisabled());

  delay(2000);
}

```

I also tried if the Hardware Serial1 of the Uno R4 is working. It is.
I can send and receive single letters.

Do you have any idea why the communication between the Uno R4 and TMC2209 is still not possible?

hefl99, 2025-03-03:
> Which TMC2209 driver board are you using?

I am using the Bigtreetech TMC2209 V1.3 driver board (https://amzn.eu/d/4pO5STx).

Here is the manual: https://device.report/m/9fc99fb405e2289eeaa464161939e06e22968f7513ec1289ef17e4965037e4d3.pdf

peterpolidoro, 2025-03-03:
Do you have voltage applied to VM or just to VIO?

hefl99, 2025-03-03:
Just VIO

peterpolidoro, 2025-03-03:
Try applying 5V to both VIO and VM and see if you can get it to communicate.

hefl99, 2025-03-04:
I tried, but still the error persists.

hefl99, 2025-03-04:
When using the library, i can not see voltage changes on the TX pin of the R4.
However, when i use other libraries that use UART on Serial1 i can see voltage changes on the TX pin of R4.
It seems, that when using the TMC2209 library the Uno R4 is not sending anything via Serial1 interface.

peterpolidoro, 2025-03-12:
I added some test code to see if it will work for your board. Are you able to pull the latest changes to this repository and test the examples/BidirectionalCommunication/TestUnoR4 example?

hefl99, 2025-03-13:
I pulled the latest commit and tried to run the TestUnoR4:
The board disconnects and doesn't reconnect unless I reboot (2x reset).
I also checked the TX pin with an oscilloscope but nothing is sent there.

I added some print statements to the setup function overloads and found that the wrong setup function is used: 
```
void TMC2209::setup(HardwareSerial & serial,
  long serial_baud_rate,
  SerialAddress serial_address)
{
  Serial.println("Wrong setup (original)...");

  #if defined(ARDUINO_UNOR4_MINIMA)
  Serial.println("ARDUINO_UNOR4_MINIMA is defined2");
  #else
  Serial.println("ARDUINO_UNOR4_MINIMA is NOT defined");
  #endif

  delay(1000);
  hardware_serial_ptr_ = &serial;
  hardware_serial_ptr_->end();
  hardware_serial_ptr_->begin(serial_baud_rate);

  initialize(serial_baud_rate, serial_address);
}
```

However ARDUINO_UNOR4_MINIMA is defined:
```
Wrong setup (original)...
ARDUINO_UNOR4_MINIMA is defined
```

I can't figure out why the wrong setup function is applied.

peterpolidoro, 2025-03-13:
I made a mistake long ago when I first created this package by calling serial "begin" in this library's setup code rather than requiring that begin is called by the user code before passing it into setup.

Anyway, I made some more changes. Try the newest version and see if it now works for you. Thanks!

hefl99, 2025-03-18:
I tested the newest version.
The correct setup overload is used.
However the command ``serial1.end();`` causes a problem.
When not commented out the program just stops there and i need to reboot the Arduino to upload any sketch again.
I also renamed serial to serial1 to avoid conflicts with Serial, which i used for debugging print statements.

serial1.end(); not commented out:
[image]
--> Program stops at serial1.end(), debug print is not executed.

//serial1.end(); commented out:
[image]
--> The Arduino is no longer freezing but is also not communicating, even if driver power is turned on (to 10V).

I probed the TX pin of the Arduino and now i can see communication at setup and during the execution of the loop. This happens only if //serial1.end(); is commented out.
[image]
[image]

hefl99, 2025-03-19:
I tried it with a completely new driver module, but still the same results.

peterpolidoro, 2025-03-19:
You mean you tried this same library with different hardware?

Have you tried your hardware setup with a different library, like:

https://github.com/teemuatlut/TMCStepper

Right now it is not clear to me whether this is a problem with this library, with your hardware setup, or both. If your hardware setup works with a different library then I want to figure out what is the difference between the two libraries that makes this one fail.

hefl99, 2025-03-19:
Yes i tried it with different hardware.

I also tried the TMCStepper Library. This one works fine. So I think my
hardware and wiring is good.

peterpolidoro, 2025-03-19:
Ok, I will look at the differences between the two libraries and try to figure out what is wrong with this library when I have a chance. Thanks!