janelia-arduino/TMC2209 #91: any suggestion on how to debug?

rmalchow, 2025-03-20:
hi,

i am trying to use this library on a seed studio xiao rp2040. this has "Serial1" and "Serial2" defined, and the pinout is:

[image]
https://files.seeedstudio.com/wiki/XIAO-RP2040/img/xinpin.jpg

i tried thes three tmc2209 boards:

[image]

all three have explicitely labelled "tx" and "rx" pins. the baudrate test always returns:

```
Stepper driver not setup and communicating!
```

(i tried both Serial1 and Serial2, as well as swapping TX and RX)

first question is - am i right in assuming this _should_ work? if yes, then what kind i do to dig into this a little bit deeper?

regards

.rm

rmalchow, 2025-03-21:
so - i have hooked the ports from the rp2040 up to a FT232R, and i am getting this byte sequence:

```
0x05 0x00 0x80 0x00 0x00 0x01 0xc0 0xf6 0x05 0x00 0x90 0x00 0x01 0x1f 0x10 0xdf 0x05 0x00 0xec 0x10 0x01 0x00 0x53 0x97 0x05 0x00 0xf0 0xc1 0x0d 0x00 0x24 0x06 0x05 0x00 0xc2 0x00 0x00 0x00 0x00 0x45 0x05 0x00 0x91 0x00 0x00 0x00 0x14 0x1f 0x05 0x00 0x93 0x00 0x00 0x00 0x00 0x0f 0x05 0x00 0xa2 0x00 0x00 0x00 0x00 0x0e 0x05 0x00 0x94 0x00 0x00 0x00 0x00 0x34 0x05 0x00 0xc0 0x00 0x00 0x00 0x00 0x8d 0x05 0x00 0xc2 0x00 0x00 0x00 0x00 0x45 0x05 0x00 0x81 0x00 0x00 0x00 0x02 0x19 0x05 0x00 0x90 0x00 0x01 0x00 0x00 0x5b 0x05 0x00 0xec 0x10 0x01 0x00 0x50 0xd9 0x05 0x00 0xf0 0xc1 0x09 0x00 0x24 0x45 0x05 0x00 0xf0 0xc1 0x01 0x00 0x24 0xe7 0x05 0x00 0x00 0x48
```

looks rather long, but, i don't know - is this ok? if i hook this up to the bigtreetech tmc2209 v1.3 on these pins:

[image]

should this theoretically work?  (of course, given that there is 3.3v and motor power applied as well)

curious

.rm

costyn, 2025-03-22:
I also cannot get this board to work. I have hooked up the Bigtreetech tmc2209 v4 to the tx/rx of an espressif esp32 devkitc tx/rx, disabled the esp32 so I'm directly receiving serial from the controller. Nothing, not a peep.

Also connecting it to the esp32 and uploading an example sketch, nothing.

The tmc board has power, 5v to the VIO pin, so I would expect to see some serial output.