janelia-arduino/TMC2209 #92: TMC2209 not working properly when starting UART mode (unidirectional)

MarcoDuttoSBM, 2025-03-21:
### Description of the environment

I need 7 stepper motors.

I'm using 7x ["BIGTREETECH TMC2209 V1.3"](https://biqu.equipment/products/bigtreetech-tmc2209-stepper-motor-driver-for-3d-printer-board-vs-tmc2208) modules on a ["BigTreeTech Octopus-Pro-V1.1"](https://github.com/bigtreetech/BIGTREETECH-OCTOPUS-Pro) mainboard.

I'm using it for an automation project different then 3D printing, so **I can't use Marlin or similar firmwares**.

I'm using Arduino (Ver. 2.3.4) IDE to write and compile the code and [STM32duino library](https://github.com/stm32duino) to compile and operate with the STM32H723ZET microcontroller from STMicroelectronics.

The OCTOPUS Pro Schematics doesn't allow for Bidirectional UART, so **I must use Unidirectional UART**.

At the moment the system works perfectly using EN, DIR, STEP pins.

### Description of the need

The Steppers are heating too much, so **I decided to use UART mode to change the TMC2209 parameters**.

### Description of the problem

If I activate the UART mode, the system no longer works as it used to work with just EN, DIR, STEP approach.
At first I tried the examples provided with this library, then I also tried several combinations of this library parameters.
Different parameters combinations provide different problems:

- Steppers always disabled
- Steppers enabled but not running
- Steppers running with noises
- Steppers too weak
- Steppers not running all together getting enabled and disabled randomly
- Random combinations of the above

Despite days of testing, I could'n find any parameter combination that makes the TMC2209s and Steppers work as they usually work without the UART mode.

Find here some chunks of the code I'm using.

Definitions:
```
const uint8_t RX_PIN = PA6;
const uint8_t TX_DRIVER0_PIN = PC4;
const uint8_t TX_DRIVER1_PIN = PD11;
const uint8_t TX_DRIVER2_PIN = PC6;
const uint8_t TX_DRIVER3_PIN = PC7;
const uint8_t TX_DRIVER4_PIN = PF2;
const uint8_t TX_DRIVER5_PIN = PE4;
const uint8_t TX_DRIVER6_PIN = PE1;
const uint8_t TX_DRIVER7_PIN = PD3;
SoftwareSerial softSerialDriver0(RX_PIN, TX_DRIVER0_PIN);
SoftwareSerial softSerialDriver1(RX_PIN, TX_DRIVER1_PIN);
SoftwareSerial softSerialDriver2(RX_PIN, TX_DRIVER2_PIN);
SoftwareSerial softSerialDriver3(RX_PIN, TX_DRIVER3_PIN);
SoftwareSerial softSerialDriver4(RX_PIN, TX_DRIVER4_PIN);
SoftwareSerial softSerialDriver5(RX_PIN, TX_DRIVER5_PIN);
SoftwareSerial softSerialDriver6(RX_PIN, TX_DRIVER6_PIN);
SoftwareSerial softSerialDriver7(RX_PIN, TX_DRIVER7_PIN);
TMC2209 stepperDriver0;
TMC2209 stepperDriver1;
TMC2209 stepperDriver2;
TMC2209 stepperDriver3;
TMC2209 stepperDriver4;
TMC2209 stepperDriver5;
TMC2209 stepperDriver6;
TMC2209 stepperDriver7;
```

Setup function:
```
void setup() {
[...]
  stepperDriverSetup(&stepperDriver0, &softSerialDriver0);
  stepperDriverSetup(&stepperDriver1, &softSerialDriver1);
  stepperDriverSetup(&stepperDriver2, &softSerialDriver2);
  stepperDriverSetup(&stepperDriver3, &softSerialDriver3);
  stepperDriverSetup(&stepperDriver4, &softSerialDriver4);
  stepperDriverSetup(&stepperDriver5, &softSerialDriver5);
  stepperDriverSetup(&stepperDriver6, &softSerialDriver6);
  stepperDriverSetup(&stepperDriver7, &softSerialDriver7);
[...]
}
```

My own stepperDriverSetup function:
```
void stepperDriverSetup(TMC2209 *stepper_driver, SoftwareSerial *soft_serial) {
  //return;
  SoftwareSerial soft_serial_ok = *soft_serial;
  stepper_driver->setup(soft_serial_ok);
  //stepper_driver->clearReset();
  //stepper_driver->clearDriveError();
  //stepper_driver->moveUsingStepDirInterface();
  //stepper_driver->setRunCurrent(100);
  //stepper_driver->setHoldCurrent(100);
  //stepper_driver->setMicrostepsPerStepPowerOfTwo(4);
  //stepper_driver->disableCoolStep();
  //stepper_driver->disableStealthChop();

  //stepper_driver->disableDoubleEdge();
  //stepper_driver->disableVSense();
  //stepper_driver->disableAutomaticGradientAdaptation();
  //stepper_driver->enableAutomaticCurrentScaling();
  //stepper_driver->disableAnalogCurrentScaling();
  //stepper_driver->enableStealthChop();
  //stepper_driver->setStealthChopDurationThreshold(300);
  //stepper_driver->setStallGuardThreshold(100000);
  stepper_driver->enable();
}
```

Considering stepperDriverSetup(...) function above:

- When I "return" before "stepper_driver->setup(soft_serial_ok);" everything works normally
- When I execute only "SoftwareSerial soft_serial_ok = *soft_serial;" and "stepper_driver->setup(soft_serial_ok);" Steppers are not working
- When uncomment also "stepper_driver->moveUsingStepDirInterface();" still motors are not running using "DIR, STEP" pins
- When I uncomment some of the "disable" lines Steppers start spinning.
- After the steppers run, the line "setMicrostepsPerStepPowerOfTwo(4);", in combination with some of the "disableXxxx" lines, seems to provide a similar behaviour (speed) to the original running without UART... but still with hickups, noises, and weird behaviours.

Any idea? Any suggestion for a list of parameters that may make the driver operate exactly like without UART mode?

Any help will be welcome!

DIY-Machines, 2025-04-15:
In would like to chime in that I am experiencing the same behaviour to help reassure you that it is not an isolated issue. I'am using the BIGTREETECH v1.3 and an ESP32.

marcodutto, 2025-04-15:
> In would like to chime in that I am experiencing the same behaviour to help reassure you that it is not an isolated issue. I'am using the BIGTREETECH v1.3 and an ESP32.

Thanks for sharing. Let's hope for some suggestion. So far I kept trying and found no solution.

MarcoDuttoSBM, 2025-07-10:
Thanks for your comment!
Regarding the first suggestion, unfortunately I already started working with just one driver.

Regarding the serial:
What do you mean by "remove all of the other SoftwareSserial setups"?

According to my tests, I think the communication is working, because when I send some commands the device reacts correctly depending on the configuration. I just can't have at the same time the StallGuard function and the motor running correctly.

Just out of my feeleings, it's more likely to be a configuration problem... like the library is setting some wrong parameters as default... or it's not setting something needed... or maybe just me ignoring how to configure the parameters. But my knowledge is very limited on this to be able to debug deeper.