janelia-arduino/TMC2209 #99: Only one of four TMC2209s is "setup", but all communicating

nhansendev, 2025-08-31:
Following the example setup for multiple drivers on one UART line with bidirectional communication, I attached four drivers to a Raspberry Pi Pico on UART2 (Pins 8 and 9).

Like so, but with four drivers and unique addresses set for each (VCCIO = 3V3):
[image]

However, I've found that at best I get just one driver to be "setup and communicating" (always address 0), while the rest only appear as "communicating, not setup". I can individually enable/disable them via software and run them with step/dir, but only the first responds to more advanced UART config, velocity control, etc.

I've attached my test script which sets-up the drivers, checks the connections, and iterates through each driver, enabling it and running it forwards/backwards, then disabling it again (triggered via pushbutton).
[TMC2209_PiPico.zip](https://github.com/user-attachments/files/22068628/TMC2209_PiPico.zip)

I've tried adjusting the reply delay, adding pauses between setup calls, different baud rates, and power cycling the MCU and driver boards, but there were no improvements.

What can I look at next to determine why only one is fully setup?

peterpolidoro, 2025-09-08:
Have you tried looking at the serial signals on a scope? I do not have a hardware setup to test this situation myself. There may be bugs in this library that prevent this from working properly or perhaps there are electrical issues with wiring it up in this way. Does it work with only two drivers connected? If you do not have any other serial ports available, then you may want to consider connecting a multiplexer like the MUX507 if you still have some digital IO pins available on your microcontroller instead.

I used a serial multiplexer with a Raspberry Pi Pico on this project:

https://github.com/janelia-kicad/cluster-pcb/tree/uart

nhansendev, 2025-09-21:
I wonder if this is at all related to why I'm having issues with Klipper communicating with the same drivers. It fails to read multiple registers like IFCNT, so maybe they have a similar issue with their protocol. I've seen various mentions of people having such issues, but no real "fix".

vijayk13, 2025-09-21:
> I've tested it on my setup and it seems to be working perfectly. Thank you for sharing this!

Good to hear!
Can't comment about the Klipper issues, but the multi-driver setup does seem to be fairly sensitive to how the UART bus transactions are managed.

Will defer to @peterpolidoro on what next steps make sense, and whether/how to bring in the (appropriate subset of the) proposed fixes into the TMC2209 library.